groupgets/purethermal1-firmware #9: Can't erase flash memory in pureThermal ST

dakkumar, 2017-02-09:
Please see the terminal-snapshot below. I can't erase pureThermal's flash memory. Please help. What might the problem be?

[image]

kekiefer, 2017-02-09:
This is the kind of error you would see if something weren't plugged in properly, or maybe if the processor didn't have power applied to it.

dakkumar, 2017-02-16:
Hi Kurt,
1. I still cannot run st-flash erase. I still see the terminal behave exactly as in the screen shot above.
2. I am following instructions as in the video clip at https://www.youtube.com/watch?v=omKEBaMih5g.
3. When I connect my ST device to my Mac, I see the power LED glow on the ST, but I measure only 0.611 V on pin J1-VDD of the ARM breakout board (ARMBB), where I expect to see 3V.
4. When I disconnect the USB cable, I can trace continuity from ARMBB-J1-VDD to my PT1-J9-1, so I know that my FFC is connected properly.
5. I see no LEDs turn on on the PT1 board because of course there's no 3V going to the PT1.

Please suggest what I might try. Have you seen/heard_of this problem happen to others as well?

closed 2017-02-16 by dakkumar