groupgets/purethermal1-firmware #18: Is there a way to build the firmware on Windows

ShuangjunLiu, 2018-08-07:
There is build instruction for Linux and Mac in readme.  
Is there a way to build the firmware on Windows?

kekiefer, 2018-08-07:
@PureEngineering put together a guide for how to install the GCC toolchain on Windows. From there, you should be able to use the Makefile to build: http://www.pureengineering.com/blog/armgcctoolchainsetuponwindows

ShuangjunLiu, 2018-08-08:
I tried the methods above but got following errors 

> ...
> usbd_uvc_if.c:(.text+0x150): undefined reference to `USBD_UVC_SetRxBuffer'
> Src/usbd_uvc_if.o: In function `UVC_VS_ControlGet':
> usbd_uvc_if.c:(.text+0x27c): undefined reference to `g_lepton_type_3'
> usbd_uvc_if.c:(.text+0x280): undefined reference to `USBD_UVC_CfgFSDesc_L2'
> usbd_uvc_if.c:(.text+0x284): undefined reference to `USBD_UVC_CfgFSDesc_L3'
> Src/usbd_uvc_if.o: In function `UVC_Transmit_FS':
> usbd_uvc_if.c:(.text+0x296): undefined reference to `HAL_NVIC_DisableIRQ'
> usbd_uvc_if.c:(.text+0x2a0): undefined reference to `USBD_UVC_SetTxBuffer'
> usbd_uvc_if.c:(.text+0x2a6): undefined reference to `USBD_UVC_TransmitPacket'
> usbd_uvc_if.c:(.text+0x2ae): undefined reference to `HAL_NVIC_EnableIRQ'
> Src/usbd_uvc_lepton_xu.o: In function `VC_LEP_GetAttribute':
> usbd_uvc_lepton_xu.c:(.text+0x2d4): undefined reference to `LEP_GetAttribute'
> Src/usbd_uvc_lepton_xu.o: In function `VC_LEP_SetAttribute':
> usbd_uvc_lepton_xu.c:(.text+0x2f0): undefined reference to `LEP_SetAttribute'
> Src/usbd_uvc_lepton_xu.o: In function `VC_LEP_RunCommand':
> usbd_uvc_lepton_xu.c:(.text+0x318): undefined reference to `LEP_RunCommand'
> collect2.exe: error: ld returned 1 exit status
> make: *** [main.out] Error 1

**Environment:** 
windows 10 
following the instruction provided in the link to install the packages. 
latest version of 
gcc-arm-chain: 
gcc-arm-none-eabi-5_4-2016q3-20160926-win32.exe

msys 1.0 

It seems to be definition missing.  Could you provide further hint on this?

closed 2018-12-04 by kekiefer