groupgets/purethermal1-firmware #19: FFC Control on Purethermal 2

Jens1503, 2018-11-26:
Hey,

I use the PT2 with the lepton 3.5 on a Mac OS, and I would like to control the shutter.
Does this repository hold the solution to my struggle? 

cheers,
jens

kekiefer, 2018-11-26:
Have you seen this? https://github.com/groupgets/purethermal1-firmware/wiki/Lepton-CCI-through-UVC-extension-units

You could take a look at GetThermal, it sends some CCI commands using these extension units. There are also some python examples here https://github.com/groupgets/purethermal1-uvc-capture

Jens1503, 2018-11-27:
Ok, so I have tried to sort this out. I m totally new at this, so I hope you could give me some hints and helps :) 

So, as I write in python I have tried to use the purethermal1-uvc-capture repository. In the uvstypes.py I added the following:

`def set_manual_ffc(devh):
            LEP_SYS_FFC_SHUTTER_MODE_MANUAL = 512 + 60 # LEP_SYS_FFC_SHUTTER_MODE_MANUAL = LEP_CID_SYS_FFC_SHUTTER_MODE_OBJ + 0 = (LEP_SYS_MODULE_BASE + 0x003C ) = 512 + 60
             set_extension_unit(devh, SYS_UNIT_ID, LEP_SYS_FFC_SHUTTER_MODE_MANUAL, 16, 32) `

Jens1503, 2018-11-27:
Ok, so I have tried to sort this out. I m totally new at this, so I hope you could give me some hints and helps :)

So, as I write in python I have tried to use the purethermal1-uvc-capture repository. In the uvstypes.py I added the following:
[image]
Where `# LEP_SYS_FFC_SHUTTER_MODE_MANUAL = LEP_CID_SYS_FFC_SHUTTER_MODE_OBJ + 0 = (LEP_SYS_MODULE_BASE + 0x003C ) = 512 + 60`

When I add this function in the uvc-deviceinfo.py and run the code, I get the following message: 
`Kernel died, restarting`

Any Idea what I do wrong?

kekiefer, 2018-11-27:
I think you're close.

You're providing [`libuvc.uvc_set_ctrl`](https://ken.tossell.net/libuvc/doc/group__ctrl.html#gaaef83f9eb6f5a2b0e4f556d2c0ec0708) with the parameter `unit` as`SYS_UNIT_ID`, then you don't need to add that offset to `ctrl`, which I think you have otherwise correct as 60. The next two parameters are `data` and `len`. `data` is the shutter position, here's the enum from the SDK:

```
typedef enum LEP_SYS_SHUTTER_POSITION_E_TAG
{
   LEP_SYS_SHUTTER_POSITION_UNKNOWN = -1,
   LEP_SYS_SHUTTER_POSITION_IDLE = 0,
   LEP_SYS_SHUTTER_POSITION_OPEN,
   LEP_SYS_SHUTTER_POSITION_CLOSED,
   LEP_SYS_SHUTTER_POSITION_BRAKE_ON,
   LEP_SYS_SHUTTER_POSITION_END

}LEP_SYS_SHUTTER_POSITION_E,  *LEP_SYS_SHUTTER_POSITION_E_PTR;
```

So `data` should be whatever you want from 0 - 3, and `len` will be in bytes, confusingly enums are all 2 bytes.

Note that if you're running a command, `data` can be anything, but `len` will be 1 byte.

I'm going to close this issue because this isn't the right place to discuss this. If you need more help, I suggest posting to the FLIR lepton google group with what you know so far (you should get more help with a better researched question) and I'll try to keep an eye on that group. Once you have the solution, posting up there will be very helpful for others.

closed 2018-11-27 by kekiefer

kekiefer, 2018-11-27:
Edit from above, from the first the link I sent.

> control_id = ((lepton_cci_reg & 0x00ff) >> 2) + 1

So `lepton_cci_reg` is `LEP_CID_SYS_FFC_SHUTTER_MODE_OBJ` (`LEP_SYS_MODULE_BASE` + 0x3c), then your `control_id` is (0x3c >> 2) + 1 = 0x10

robyerevan, 2020-10-09:
I have same similar issue, I need to disable auto shutter from Python.
`def set_manual_ffc(devh):
    set_extension_unit(devh, SYS_UNIT_ID, 0x10, 0, 1)`

This is the code I use and it is crushing with **Fatal Python error: Segmentation fault**
@Jens1503 did you manage to fix this ?

Thank you