groupgets/purethermal1-firmware #20: how to get unique serial number for multiple thermal cameras?

shubhamwagh, 2018-12-04:
I have a setup where I require 4 thermal cameras. But all of these cameras have same serial number. May I know the changes to be made in the firmaware code in order to get unique serial number for each device?

I think the serial number is generated from the following code snippets

`#define USBD_SERIALNUMBER_STRING_FS     "v1.0.0"`

`uint8_t *  USBD_FS_SerialStrDescriptor( USBD_SpeedTypeDef speed , uint16_t *length)
{
  if(speed  == USBD_SPEED_HIGH)
  {    
    USBD_GetString (USBD_SERIALNUMBER_STRING_FS, USBD_StrDesc, length);
  }
  else
  {
    USBD_GetString (USBD_SERIALNUMBER_STRING_FS, USBD_StrDesc, length);    
  }
  return USBD_StrDesc;
}`

kekiefer, 2018-12-04:
You're right, we're hacking the usb serial number descriptor to display the version. Changing that should do the trick. There's a serial number you could get from the camera itself and also a UUID you can get from the STM32 (HAL_GetUIDw*, there may be other ways to access this), either would do the trick.

closed 2018-12-04 by kekiefer

kekiefer, 2019-05-07:
For anyone searching closed issues, the PureThermal board now has a unique per-device serial number available in the USB serial number descriptor, since firmware version 1.2.0 released earlier this year.

kerio92, 2019-06-12:
Would it be possible to also add the serial number of the Lepton module itself to the serial number? Perhaps leaving it out before the module is detected, and filling it in after the re-enumeration? It would also be a simple way of detecting a badly-inserted module.