groupgets/purethermal1-firmware #24: PT2 not working in RPi4 (4GB) with Raspbian Buster

agjunyent, 2019-08-02:
I've trying to obtain images on a newly baked RPi4 with Raspbian Buster from a PT2 board that was working perfectly in a RPi3B+, but I'm totally unsuccessful.

I've seen that whenever I try to get a frame from the camera, using any method, I get the same error in dmesg:
```
usb 1-1.3: Not enough bandwidth for new device state.
usb 1-1.3: Not enough bandwidth for altsetting 1
```
Whether if I connect the board to a USB2 port, or to a USB3 port, the error is always the same.

I've tried with Python OpenCV, ffmpeg, guvcview, and your examples with other means, but I'm always getting the same error.

I started to play with uvcvideo quirks, but still no success at all. 

Any help will be appreciated!

kekiefer, 2019-08-20:
I haven't heard of anything but problems with this platform. So I have just ordered one for myself to see what I can see.

kekiefer, 2019-08-23:
It seems like this is a problem with the firmware in the Via USB controller on the RPI and how much bandwidth it's willing to dedicate to isochronous endpoints on a full speed device. The PT asks for 962 bytes per frame at 1000 frames/sec, which should be perfectly valid. However, at some lower bandwidth, it will accept the streams.

The following patch will work around the limit for 16-bit frame types (Y16 and 422, not RGB), but it won't work as a general solution without some more effort.

```
From: Kurt Kiefer <kekiefer@gmail.com>
Date: Thu, 22 Aug 2019 21:06:57 -0700
Subject: [PATCH] HACK: lower bandwith for stream compatibility

---
 .../ST/STM32_USB_Device_Library/Class/video/Inc/usbd_uvc.h      | 2 +-
 1 file changed, 1 insertion(+), 1 deletion(-)

diff --git a/Middlewares/ST/STM32_USB_Device_Library/Class/video/Inc/usbd_uvc.h b/Middlewares/ST/STM32_USB_Device_Library/Class/video/Inc/usbd_uvc.h
index 8200aa8..ed38c91 100644
--- a/Middlewares/ST/STM32_USB_Device_Library/Class/video/Inc/usbd_uvc.h
+++ b/Middlewares/ST/STM32_USB_Device_Library/Class/video/Inc/usbd_uvc.h
@@ -60,7 +60,7 @@
 
 #define UVC_IN_EP                                     0x81  /* EP1 for data IN */
 #define UVC_CMD_EP                                    0x82  /* EP2 for UVC commands */
-#define VIDEO_PACKET_SIZE                             ((unsigned int)(962))
+#define VIDEO_PACKET_SIZE                             ((unsigned int)(482)) /* min 162 for L2 Y16, 482 for L3 Y16 */
 #define VIDEO_MAX_SETUP_PACKET_SIZE                   ((unsigned int)(1024))
 #define CMD_PACKET_SIZE                               ((unsigned int)(8))
 
-- 
2.21.0
```

agjunyent, 2019-08-23:
This seems interesting. I've opened an issue in RPi repo (https://github.com/raspberrypi/linux/issues/3136) also, because I thought is was a firmware issue.

Could you take a look and comment there also? This is driving me crazy...

agjunyent, 2019-08-23:
Can confirm this patch works with Y16 at least, that is the format that I'm using with Python and OpenCV to get the images.

Thanks. I understand that the problem in on the RPi end, on the USB firmware or kernel when trying to reserve bandwidth. I'll post this patch on their issue too, maybe it gives a clue to them on how to solve the problem in their end.

Thanks @kekiefer

longo92, 2019-11-10:
Hi,
i have the same driver issue on an android device, then i choose to use your patch and seems work (i don't have error ENOSPC in usb_submit_urb). 
I use an L3 at 160*120@9 fps with pixel format UYVY and your patch applied in release 1.2.2 ( i change only #define VIDEO_PACKET_SIZE  482) but i get this image distortion:
[image]
The image distortion is also presente if i use the PTmini on a computer, using CamView(http://www.penguin.cz/~fojtik/CamView/).
Do you know possible reasons?

kekiefer, 2019-11-11:
Can you try with the latest code on development and see if it works better?

longo92, 2019-11-11:
With latest commit on development branch works on computer.
On the android device i get the same ENOSPC error, it seems 642 si too much.

jaimegiga, 2019-11-25:
Hi,

I have the same issue with Purethermal1 and Lepton3.5 in the RPi4. In order to apply the patch suggested before, is it necessary to use the ARM MCU JTAG Adapter and ST-Link/V2? I updated the firmware to latest release version via DFU-Mode in Windows. I would like to confirm the correct way  to carry out the modifications.

Thank you in advance.

kekiefer, 2019-11-25:
Some PT1 boards can do DFU better than others, they don't all work reliably. If the flash command completed with success, this is OK.

jaimegiga, 2019-11-28:
GetThermal and uvc-radiometry.py working great after the patch on my PT V1.3 in RPi4!

Thanks @kekiefer

kekiefer, 2019-11-28:
Good to hear. Was this the patch, or the code on the development branch?

I'd like to merge the code from development and stamp a new release, but I believe you might also have to upgrade libuvc to the latest upstream (https://github.com/libuvc/libuvc) and then also the `master+libuvc-upstream` (https://github.com/groupgets/purethermal1-uvc-capture/tree/master+libuvc-upstream) branch of the examples.

jaimegiga, 2019-11-28:
I built the firmware from the development branch.

kekiefer, 2019-11-28:
OK, great. That makes the upgrade path much easier. I'll be merging this code and making a new release soon.

closed 2020-05-08 by kekiefer