groupgets/purethermal1-firmware #26: Control of AGC etc. on Bananapi M2 Zero

ystumtec, 2019-11-04:
Hi, 
I've got a system using a PureThermal-M running on a bananapi M2 Zero which works fairly well after a bit of playing around. I am running it from a C program based on the grab example using the Y16 format. I can capture images etc... but I would like to have control over the gain and exposure parameters. I guess this is just about knowing which IOCTLs to use, but I'm not having much joy hunting down any examples/documentation. Slightly worryingly I get the following (and many more errors) when running "v4l2-ctl -l" after setting up uvcdynctrl.

**error 22 getting ctrl Brightness
error 22 getting ctrl Contrast
error 22 getting ctrl LEP_CID_AGC_ENABLE_STATE
error 22 getting ctrl LEP_CID_AGC_POLICY
error 22 getting ctrl LEP_CID_AGC_ROI
error 22 getting ctrl LEP_CID_AGC_STATISTICS....**

I'm not too worried about getting the command-line tools running, but I'm guessing that these errors are symptoms of a more fundamental underlying problem with my setup.

It may have something to do with the kernel that I'm running:
Linux bpi-iot-ros-ai 3.4.113-BPI-M2Z-Kernel #7 SMP PREEMPT Thu Jan 3 16:33:44 CST 2019 armv7l GNU/Linux
but I'd really like to avoid upgrading to a version 4 kernel if I can as the bananapi is a bit less well supported than I'd like and it's taken a while to get everything set up. It was the only small form-factor board with sufficient performance that I could find/was aware of for my application.

Any hints/pointers to examples of IOCTL use for this gratefully received!

Thanks,

Mark.

kekiefer, 2019-11-04:
All I can say about this is that error 22 is a timeout. I have no idea why this would happen on your platform, especially just probing some of the controls. Is it power maybe? Could you try attaching the camera to a powered USB hub just for test?

kekiefer, 2019-11-04:
Oh, another thing you could try if you think this is just a kernel driver issue affecting the V4L2 controls is bypass the kernel in this. See the python examples here that send some UVC extension unit commands with libuvc: https://github.com/groupgets/purethermal1-uvc-capture

ystumtec, 2019-11-04:
Thanks for the quick responses!

Mmm, interesting. I had just assumed that it was a driver/versions problem. The camera will capture images perfectly well so it seems pretty unlikely to me to be a power problem, but stranger things have happened. I also get error 22 when I try to use formats other than BGR24 or Y16. It is also completely predictable. I will try a powered hub at some point, but it means rather a lot of dismantling.

I guess you mean the calls to "uvc_get_ctrl" and "uvc_set_ctrl". I saw this in libuvc documentation here as well: 
https://ken.tossell.net/libuvc/doc/group__ctrl.html#ga185b4c3cc654690d7cd6271b67245561

But I'm confused about where I can find a list of the unit and control parameter values.  Do these map somehow to the information in the pt1.xml file?

Sorry if these are naive questions!

kekiefer, 2019-11-04:
This may help https://github.com/groupgets/purethermal1-firmware/wiki/Lepton-CCI-through-UVC-extension-units and also have a look at the flir lepton google group.

ystumtec, 2019-11-04:
Ah, fantastic. Some real information to chew on.

I'll write some code to sidestep the driver stuff and see if that helps. 

Thanks.

closed 2019-11-11 by kekiefer