groupgets/purethermal1-firmware #27: Lepton STM32 firmware incomplete

bobmc-rmm, 2019-11-06:
Starting with PT version 1.2.2, I am trying to set/get data from a region-of-interest ROI. Converting from RGB to YUV and streaming over UVC to an application computer is obvious in usb_task. However, the working of UVC is a mystery. For example, in usbd_desc.c there are functions to pack descriptors but they are not linked in the source. In usbd_uvc_if.c, UVC_Receive_FS() does nothing but return OK.

Please tell me how I can implement a UVC data flow for ROI settings and measures.

Using https://www.perceptsystems.ca/netthermal

kekiefer, 2019-11-06:
Is this a question about NetThermal? If so, this is not the right place to ask questions about that.

If you're looking for examples on how to send arbitrary Lepton SDK commands you can see here https://github.com/groupgets/purethermal1-uvc-capture and here https://github.com/groupgets/purethermal1-firmware/wiki/Lepton-CCI-through-UVC-extension-units

bobmc-rmm, 2019-11-06:
I visited those pages before writing this issue about the firmware.  I would like to understand how the current design works before I start tinkering.  Looking at the code gives me more questions then answers. An API or a working example of parameter setting and getting would suffice.

kekiefer, 2019-11-06:
Well, we use the Lepton ROI features in https://github.com/groupgets/GetThermal with the PT boards. Other than pointing you to these things and to the Lepton IDD, I'm not sure how else to help you.

bobmc-rmm, 2019-11-06:
Ok, I rest for now.

regards

closed 2019-11-06 by bobmc-rmm