groupgets/purethermal1-firmware #28: Possibility to trigger firmware restart from host?

ghellwig, 2019-12-09:
Hi all,

I have a question regarding stability of a long running application.
I got the impression that aborted streaming or extension unit requests leave the firmware on the board in some inconsistent state.
This leads to a non-responsive camera which does not provide frames anymore.
Obviously, one should try to avoid the above-mentioned aborts, but I would like to know if it is possible to send some kind of restart request instead of unplugging the board from the host.

Thank you for your help,
Gregor

ghellwig, 2019-12-11:
To put this into context: I am using the camera in some continuous streaming mode. When doing so, the camera stops streaming at some point and is irresponsive. This can only be cured unplugging the camera and thereby triggering a firmware restart.
Obviously it would be better to fix the cause of the irresponsive state, but I would like to know anyways.

ghellwig, 2019-12-11:
@kekiefer @Sheyne 
I think, you two are most likely the best candidates to answer my question.

I found the command ID [LEP_CID_OEM_REBOOT](https://github.com/groupgets/purethermal1-firmware/blob/master/Middlewares/lepton_sdk/Inc/LEPTON_OEM.h#L99), but issuing this run command leaves the camera in an irresponsive state. In addition it looks like the power_on command is not available via XUs. This might have to do with it to be not fully supported as the documentation states, but I am not sure about this, though.

ghellwig, 2019-12-11:
@Sheyne 
It looks like your PR https://github.com/groupgets/purethermal1-firmware/pull/14 introduced a feature which might be useful for me in order to power_on the camera, right?
Can you provide some usage instructions?
Thank you, for your help!

ghellwig, 2020-01-02:
Happy New Year!
As I have seen some activity from you, @kekiefer, I got reminded of this issue here.
Could you give me a hint on this one?

kekiefer, 2020-01-02:
Sorry @ghellwig I don't know how to do this. For one thing, there is a bit of setup that the firmware does at startup which will have to be repeated (most importantly enable the vsync gpio, also set modes and color palettes), but if the thing is totally unresponsive you won't get this far. There may be some threads about this on the google group that might have more information.

ghellwig, 2020-01-02:
Hi @kekiefer, thanks for your feedback.
But do you have an idea on how to restart the whole firmware?
Btw, which exact Google group are you referring to?

kekiefer, 2020-01-02:
I wouldn't restart the whole firmware, since there are other initialization functions that are performed that maybe couldn't be repeated. You can step through and see what it's doing to the Lepton, those would we the things you'd repeat. Here is the google group: https://groups.google.com/forum/#!forum/flir-lepton

ghellwig, 2020-01-06:
Ok, thanks for the link. I'll have a look.
Regarding the steps to perform: Is it correct that you are referring to these steps:
https://github.com/groupgets/purethermal1-firmware/blob/master/Src/lepton_task.c#L122-L169

I am also wondering if it could be that the lepton task died for some reason, e.g. unhandled exceptions, etc.
In this, I guess, a camera restart is rather useless, right?

ghellwig, 2020-02-10:
Hi @kekiefer,
I have a quick question regarding this issue.
From a [discussion in the Google group](https://groups.google.com/forum/#!searchin/flir-lepton/reboot/flir-lepton/z_BufL8zeOE/fLnEh2Q8AwAJ) I understand that it is sort of normal to obtain a `LEP_RESULT` of `LEP_ERROR_I2C_FAIL` when issuing `LEP_RunOemReboot()`.

So, I was wondering if it would make sense to implement a special handling of the command ID `LEP_CID_OEM_REBOOT` within [`LEP_I2C_RunCommand_PT()`](https://github.com/groupgets/purethermal1-firmware/blob/master/Src/lepton_i2c_task.c#L290-L359).
However, I have no idea how exactly and if one needs to take into account other dependent tasks which become invalid while rebooting the camera.

Any suggestions?

ghellwig, 2020-02-10:
Basically, I want to achieve the same thing as the RST button of the PureThermal Mini is doing...

kekiefer, 2020-02-10:
You can try calling `HAL_NVIC_SystemReset()` -- this will generate a system reset, and the code will start over from the reset vector like on POR from the reset button. There could be problems reinitializing clocks and peripherals for a second time if they're not shut down properly first, but it's certainly worth a try.

ghellwig, 2020-02-11:
Thank you, I'll give that a try!

ghellwig, 2020-02-11:
Just one thing, I noted: There is a special handling of FFC run commands here:
https://github.com/groupgets/purethermal1-firmware/blob/master/Src/lepton_i2c_task.c#L675-L676

What's the rationale behind this?
Is it required to do this for certain run commands?

ghellwig, 2020-02-11:
`HAL_NVIC_SystemReset()` seems to work as expected, i.e. the board performs a restart.

Although this works, it is of course a dirty hack as you probably agree.

My observations with my previous attempts were the following:
* checking for `LEP_CID_OEM_REBOOT` in `LEP_I2C_RunCommand_PT()` is quite easy
* putting a `HAL_Delay(1000)` after the actual run command is sufficient to perform the reboot
  - I infer this from the fact that the shutter closes
  - I can repeat this command with the same effect, i.e. the `PT_THREAD` `lepton_attribute_xfer_task` is still in a valid state
* the `PT_THREAD` `lepton_task` seems to fail as streaming has stopped
  - I am wondering if it is possible to cleanly trigger a restart of this thread
  - I played around a bit with `PT_RESTART` in the while loop within the while loop of `lepton_task`, but this was insufficient

Do you have any idea what I need to do cleanly restart `lepton_task`?

closed 2020-03-26 by ghellwig

Steve7522, 2021-08-30:
@ghellwig 
Hi ghellwig
I have the same problem as you but still do not find any solution so far. Do you solve the question of the non-responsive camera which did not provide frames anymore. Can you teach me to do next steps? I have looked for many ways, but I still can’t find a solution to finish it.

Thx