groupgets/purethermal1-firmware #29: select timeout error

ajal-fh, 2019-12-19:
I'm currently running two lepton 3.5 cameras using two purethermal mini boards. I use opencv for continuous streaming of images.  One of the boards runs continuously without any errors (tested for 2hrs). The other board gives 'select timeout' error after an hour (As i have tested multiple times the duration for which it runs without problems varies). As I could understand this timeout error occurs when there is no data send from the usb video device. I have also replaced the lepton camera on this purethermal mini and still the issue persists. Any help would be much appreciated.

tested on 
Raspberry Pi 3b+ (Debian stretch, OpenCV 3)

Regards
Ajal

reymarkdivino, 2020-02-02:
Hi @ajal-fh

I've encounter the same problem. did you find any solution ?

Thanks!

ajal-fh, 2020-02-10:
> Hi @ajal-fh
> 
> I've encounter the same problem. did you find any solution ?
> 
> Thanks!

I'm sorry I couldn't resolve the issue.

jarvis-owl, 2020-05-06:
Same here with Opencv 3 C++ and Python3.6.

Ubuntu 16.04 with Intel I7 NUC

bigface0202, 2020-06-09:
I'm also facing same issue on Ubuntu 20.04, intel i7.
The system detected the Lepton camera(Lepton 1.6) correctly, but cannot access the camera view.
I tried in python. Anybody find the solution?

sebasgra39, 2021-01-27:
Hey guys,

I think this has something to do with the uvcvideo settings in the linux kernel. May you have a look here and maybe report your findings?

# https://github.com/groupgets/purethermal1-uvc-capture/issues/44

BR
Sebastian