groupgets/purethermal1-firmware #30: Raspberry 3B+ with purethermal 2 failed

imostlove17, 2019-12-30:
I am a novice. And a lepton 3.5 installed on purethermal 2, my Raspberry pi 3B+ installed Raspbian Buster.
I tried to run this program: uvc-radiometry.py
I got this error message :

$ sudo ./uvc-radiometry.py
unsupported descriptor subtype VS_COLORFORMAT
unsupported descriptor subtype VS_COLORFORMAT
unsupported descriptor subtype VS_COLORFORMAT
unsupported descriptor subtype VS_COLORFORMAT
unsupported descriptor subtype VS_COLORFORMAT
device opened!
Version gpp: 3.3.26 dsp: 3.3.26
FLIR part #: 500-0771-01
FLIR serial #: '{"(\x00\x00\x00\x00\x00'
format: UYVY
frame 160x120 @ 9fps
format: Y16
frame 160x120 @ 9fps
frame 160x122 @ 9fps
format: Y8
frame 160x120 @ 9fps
format: RGBP
frame 160x120 @ 9fps
format: }�6�
frame 160x120 @ 9fps
uvc_start_streaming failed: -2

please help me.

kekiefer, 2019-12-30:
Does this happen with the firmware on the development branch?

kekiefer, 2019-12-30:
This is not a purethermal firmware issue, it looks like you may not using the groupgets fork of libuvc as per the instructions in the capture examples.

If you want to use the upstream libuvc, you can give the branch [master+libuvc_upstream](https://github.com/groupgets/purethermal1-uvc-capture/tree/master+libuvc-upstream).

closed 2019-12-30 by kekiefer