groupgets/purethermal1-firmware #32: version is displayed incorrectly

zarpli, 2020-01-20:
Hi,

I am update PT2 boards with 1.2.2 firmware, but now the version is displayed incorrectly (apparently).

please see the attached images,

regards.

[image]

zarpli, 2020-01-20:
[image]

kekiefer, 2020-01-20:
This was being handled incorrectly by the firmware prior to this release, and these applications will need to be updated (I think GetThermal has the change in git, but I'm not sure there is a binary build available).

The version number was being put in the serial number descriptor, but a024ab4cf61498640a80871ce57b1551c19fb198 added a UUID based on the ROM in the MCU that made sense to put into the serial number descriptor. The version name was moved to the product name descriptor.

closed 2020-01-20 by kekiefer