groupgets/purethermal1-firmware #33: FW Version Format

IeiuniumLux, 2020-02-17:
@kekiefer 

Can the FW Version follow the former format (e.g. v1.3.0) instead of the hex numbers?

[image]

[image]

kekiefer, 2020-02-17:
Old versions of firmware incorrectly used the camera serial number field to display the version number. This is now fixed, and you'll find the correct version by looking at the USB descriptors. So what you see is now a problem with older versions of GetThermal. In pr. The newest versions should have everything straightened out.

closed 2020-02-17 by kekiefer

IeiuniumLux, 2020-02-17:
Hmm, gotcha.

**BEFORE**:
...
  iManufacturer:           1 GroupGets
  iProduct:                2 PureThermal 1
  iSerial:                 3 v1.1.0

**NOW**:
...
  iManufacturer:           1 GroupGets
  iProduct:                2 PureThermal (fw:v1.3.0)
  iSerial:                 3 8003800e-5119-3536-3432-303200000000