groupgets/purethermal1-firmware #34: PureThermal mini without button - How to update firmware

Sergio-, 2020-12-11:
Hello. Good day.

I have this PureThermal mini and I'm wondering how can I put it in DFU mode to I can upgrade firmware.
I know it is different from the Groupgets one, Just wondering if there's any way to put it in DFU via I2C or UART. It runs the same firmware.
Thank you.

[image]
[image]

imaurus, 2020-12-16:
[image]

Sergio-, 2020-12-17:
Thanks @imaurus. It worked.
May I ask how do you reached the conclusion that this particular line needed to be commented in order of it to work on this board?
I'm looking for sharing the learning/investigation process that leaded to that conclusion.
Thank you.

imaurus, 2020-12-17:
To be honest I found this image that a guy posted on a comment of Aliexpress. Then I digged and I found that this guy created a case for this PT clone. Link: https://www.thingiverse.com/thing:4571621

closed 2020-12-17 by kekiefer

kekiefer, 2020-12-17:
Hey guys, if there's more discussion needed on the support for these third party boards, please move it to the Google Group. Someone just opened a question about these boards.

To be clear these vendors are within their rights to use this firmware (it's MIT licensed), but they do so without contributing any support back to the project, so I'm not really interested in providing any support back to them.