groupgets/purethermal1-firmware #36: How can we run 2 or more purethermal devices together ?

VegetarianLion258, 2021-01-27:


mrspirytus, 2021-05-08:
I am interested in this as well

jaysonlarose, 2022-02-27:
I was coming here to report that I'm getting this error when I try to capture video from a PureThermal Mini and a PureThermal Mini Pro JST-SR attached to the same USB3 hub under Linux:

```
[675323.979500] usb 1-4.1.3: Not enough bandwidth for new device state.
[675323.979506] usb 1-4.1.3: Not enough bandwidth for altsetting 1
```

I presume this is related to the problem @VegetarianLion258 is having.

I know just barely enough about USB HID report descriptors to be a danger, but I know absolutely nothing about UVC.... but could it be possible that these devices were misconfigured to claim to use a lot more bandwidth than they actually do?