groupgets/purethermal1-firmware #37: Questions about PureThermal 2 +LEPTON3.5

38293312, 2021-03-14:
I use PureThermal 2 +LEPTON3.5 windows Lepton User App to use it normally, but the Android system cannot display it using OTG. I used PureThermal 1+LEPTON3.5 again, the same problem, PT2 firmware is 1.3, PT1 is version 1.0. Android APP "thermal Camera Viewer". Android phones support OTG function, I also replaced several different brands of mobile phones and tablets Tests can't be used either. Hope to help me! Thank you!

nsknyc, 2021-03-16:
I'm still trying to figure this out myself. I'm currently browsing around the firmware code to see whats up (interested also in uart). Is there a group forum for this?

38293312, 2021-03-17:
> 我仍在尝试自己解决这个问题。我目前正在浏览固件代码，以查看最新情况（对uart也感兴趣）。是否有一个团体论坛？

Sorry, I’m just an electronics enthusiast. I don’t know much about the codes and programs. I just want to use PT2 on mobile phones. The official website advertises that mobile phones can be used, so I bought it, but it didn’t work. I tried it. Change to a lower version of the firmware, the result is still not working.

nsknyc, 2021-03-28:
I wanted to give you a heads up. Peter-St is trying to add uyvy support on android-uvc-camera. Here's the link.
[https://github.com/Peter-St/Android-UVC-Camera](url)
You can use my settings for now to "play" with purethermal. Check [https://github.com/Peter-St/Android-UVC-Camera/issues/18](url)
It doesn't look perfect, but we will get there soon.

38293312, 2021-03-30:
> 我想提醒你。Peter-St试图在android-uvc-camera上添加uyvy支持。这是链接。
> [https://github.com/Peter-St/Android-UVC-Camera](url)
> 您现在可以使用我的设置来“玩”纯热游戏。检查[https://github.com/Peter-St/Android-UVC-Camera/issues/18](url)
> 它看起来并不完美，但是我们很快就会到达那里。

Thank you for your help. Have you tested it with PT2+LEPTON3.5 using APP "UYVY_support.apk"? I downloaded the app you mentioned, and it failed!

nsknyc, 2021-03-30:
The author is currently working on this. We're trying to figure out the best approach. If you select YUY2 you should see something.

38293312, 2021-03-30:
> 作者目前正在对此进行研究。我们正在尝试找出最佳方法。如果选择YUY2，您应该会看到一些东西。

failure！

[image]
[image]
[image]

Peter-St, 2021-03-31:
@38293312 

> I upgraded the PT2 to the 1.3 firmware, and it failed. Here is the test video: 
> https://www.youtube.com/watch?v=3SFKlTRe-G0&t=15s`

On your video it looks like, there is a problem with the cable, OR there could be a problem with the termal cam itself. The usb cam is powerd up --> the app recognizes, that there is a camera present. Suddenly after 1 sec, the granting permissions screen disapears. You couldn't even click on the button.
This could be due to hardware defects, or defective cables, or...
Otherwise it seems you can use your cam on a pc.
Eventually changing the cable could help. Using an OTG cable with additionally power supply for example.

So far,

Peter

38293312, 2021-04-01:
> @ 38293312
> 
> > 我将PT2升级到1.3固件，但失败了。这是测试视频：
> > [https：//www.youtube.com/watch？v = 3SFKlTRe-G0＆t = 15s`](https://www.youtube.com/watch?v=3SFKlTRe-G0&t=15s%60)
> 
> 在您的视频上看起来好像是电缆有问题，或者端子凸轮本身可能有问题。USB摄像头已启动->应用程序识别出存在摄像头。1秒钟后，“授予权限”屏幕突然消失。您甚至无法单击该按钮。
> 这可能是由于硬件故障，电缆故障或...引起的。
> 否则，您似乎可以在PC上使用凸轮。
> 最终更换电缆可能会有所帮助。例如，使用带有额外电源的OTG电缆。
> 
> 至今，
> 
> 彼德

I have used several additional OTG cables. The external power supply is supplied to the OTG. All the cables have the same problem.

Peter-St, 2021-04-08:
Have you already tried an other android device?

The resent update of my app in the Play Store already supports UYVY.

Peter

ilie321, 2022-08-07:
https://github.com/ilie321/thermal-cam-android-for-purethermal-1
i made this ; maybe it is working for others but not sure (only 80*60)

Peter-St, 2022-08-07:
hi,

can you eventually try out my app again and post a sceenshot or video rec. of the screen?
I made some changes and maybe it could work:

https://github.com/Peter-St/Android-UVC-Camera

 https://play.google.com/store/apps/details?id=humer.uvc_camera&hl=de

Thx, Peter

kubabuda, 2023-01-11:
@Peter-St  I just tested your Android-UVC_Camera (latest version from Play store), with board firmware v1.2.2 and Lepton 3.5 module. 
I did some some parameters tweaking and came up with these:
[image]
and now I can see something:
![obraz](https://user-images.githubu
[image]
sercontent.com/4533714/211817812-91b4b774-2eca-4773-8ba0-560a2b340b75.png)
Camera feed is distorted: stretched to fill screen, and super glitchy. Is it matter of wrong configuration values? On laptop camera feed is stable in all the apps, even with the same OTG cable on USB C port, so cables are probably OK

Peter-St, 2023-01-11:
Hoi,

Probably it's because of the wrong values. Try to raise the PacketsPerRequest and ActiveUrbs to 4 (max. to 16).

Theoretically, if you look at the frame size, when you perform a testrun under the Setup Method you will see the size of your frames. Your Frames should have a size of (160x120x2) 38400 bytes each. If the size differs, try to edit the values for PacketsPerRequest and ActiveUrbs.

Regards, Peter

kubabuda, 2023-01-12:
I raised PacketsPerRequest and ActiveUrbs to 4. Now image is now stable - great thanks.
[image]

Last thing - aspect ratio is messed, camera feed is stretched to fill display. Is there a setting to keep original aspect ratio?

Peter-St, 2023-01-12:
👍

I think there is no option for that included up to now. Perhaps there will be one in the future.

Peter-St, 2023-01-13:
Fix for aspect ratio is available on the github repo. (new Option included)
On Play Store the new update will be in a few days.

kubabuda, 2023-01-16:
[image]
New option is working as intended

ilie321, 2023-01-16:
[image] 
My image with 80x60 camera :)