groupgets/purethermal1-firmware #39: Video freez for a second and the shutter opens and closes.

spajder16, 2021-04-14:
Hi, 
I have the lepton 3.5 and purethermal mini. No matter what software I use it with, the video from the camera freezes every now and then for a second and the shutter of the camera closes and opens.

agjunyent, 2021-04-14:
Tihs is not realted to the software, but the firmware of the camera.

The ambient and the internal temperature of the FLIR Lepton 3.5 affects the precision ,of the measurments that the Radiometric sensor gets. Every now and then, or when the drift of measurements is too high, the shutter closes, the camera recalibrates, and opens again. This freezes the video for 2-3 seconds.

You can deactivate this feature by setting the autocalibration to false, but you will get a large drift of Radiometric values over time.

spajder16, 2021-04-14:
Thank you very much for the explanation.

closed 2021-04-14 by spajder16