groupgets/purethermal1-firmware #44: Flire Lepton 3.5 on PT2 Board - Raspi 4B - can't get uvc_radiometry.py working

boydy33, 2022-05-02:
Hi. I have a Flir Lepton 3.5 on a PureThermal 2 Board and am trying to download radiometric images down to a Raspberry Pi 4B using the uvc_radiometry.py program which in turn uses the libuvc tool.
I'm really struggling to get anywhere with it and I'm yet to find an online solution. The current problem is that it seems that the raspi 4 can't see the lepton 3.5 / PT2 board when I plug it into a USB port. 
I get a res "uvc_find_device error" which basically means its returning an error on the command; res = libuvc.uvc_find_device(ctx, byref(dev), PT_USB_VID, PT_USB_PID, 0)
I found a few threads on putting in hex values instead of PT_USB_VID, PT_USB_PID but none of them worked.

Any ideas out there? As for other users trying to work around this, it does tend to drive you mad, particularly as Flir and GroupGets have failed to help me at all. Thanks in advance. Nick

DJSOX, 2022-06-28:
I use pascal (Lazarus) with RPi4 with a Lepton 3.5 using V4L works really well. So the Purethermal board/lepton do work well on the RPI4, sorry but I don't do Python.
Dan

boydy33, 2022-06-29:
Thanks for letting me know about that Dan,

As this is a commercial product on developing, I’ve moved on to a Flir A50 thermal camera now which comes with its own different challenges on a raspberry pi.

I’d be interested though to see how you did that so are you able to send me the code or is it available on GitHub or somewhere?

Cheers,

Nick