groupgets/purethermal1-firmware #48: PureThermal 2 on Android Chrome

rvegele, 2022-11-29:
I'm trying to run the purethermal module on an android device using Chrome through USB. 
It successfully shows up on a laptop/pc in chrome yet android is not showing the camera as an option to choose from using MediaDevices.enumerateDevices().

The camera does show up in chrome// devices.

I've created a bug-report on chrome dev boards, but I'm hoping it's something that I can solve, perhaps with a small firmware update myself?

kubabuda, 2022-12-29:
What purethermal firmware version are you using?

siamaktavakoli, 2023-10-22:
Hi, is this issue addressed?
I have the same issue, also trying to fix it via firmware update. Need to find where camera permission is communicated with Android 10 device? or what else?