groupgets/purethermal1-firmware #56: Issues running purethermal1 on pi zero 2w

techmaticsnz, 2023-01-25:
Hello,

We've been using your thermal board for some time now so just wanted to thank you for the continued support that's been given over using this particular product in our line of robots.

I'm experiencing a stressful issue at the moment trying to stream the purethermal1 over wifi with flask. (flir lepton 3.5)

If I start up the program I get this error (over and over)
`[ WARN:0@11.195] global /tmp/pip-wheel-nz49_ez6/opencv-python-headless_488dea44cb7d406baf80df0bfc388ff9/opencv/modules/videoio/src/cap_v4l.cpp (1013) tryIoctl VIDEOIO(V4L2:/dev/lepton_camera): select() timeout.`

I may get a single frame to display every minute.

The board is just running one purethermal device.

We've tried using the product, same program, on a pi3b, also a pi zero, and a pi 4.

We only have access to one pi zero 2, so are looking for an alternative solution to determine how to fix than simply replacing pi zero 2  (as you are aware limited numbers world wide)

What should we be using and is there an alternative way to stream the device over wifi?

Regards

Nathan

techmaticsnz, 2023-01-27:
I've done a little digging. Can I get some help with what could possibly be happening?

This is the dmesg log when running the opencv program.
https://pastebin.com/FeypS204

I've tried using libuvc version of getting frames and found out some more helpful information.
using this program:

https://pastebin.com/RUAkVxEJ

it appears that after 30 seconds it can't find any frames from the callback and closes.

Tried camera on a pi zero, Pi3b, and Pi4b. Tried the camera on the same program on Pi3b and Pi4b, works.
It appears to be something specific to the pi zero 2.
It appears to be an issue with the start_streaming function.

Should note that the opencv program DOES receive frames every once in a while. but it is largely the select timeout issue trying to read frames from the camera.

techmaticsnz, 2023-01-27:
It doesn't appear to be related to hardware... We tried plugging in a keyboard to the USB port and it works fine.
Also if we run the camera through a USB hub it actually starts getting data, but after quickly after starting the program it closes because it doesn't read any frames.

EduardoRonchi, 2023-03-16:
Any updates on this issue? I am having the same problem with PT Mini + Lepton 3.5 and Raspberry Pi 4B. If I restart my raspberry solves the problem, but it comes back after some time.

techmaticsnz, 2023-05-25:
Sorry for the late reply, but we've found that the Purethermal Breakout board a good replacement for the Purethermal 2. It connects the thermal camera over I2C and SPI. Hope this still helps