groupgets/purethermal1-firmware #57: purethermal2 and android device

idan123456, 2023-02-23:
Hey, I'm trying to create an android app that previews the purethermal image,
And when I try to enumerate the connected cameras in android using Camera manager it doesn't list the purethermal. Still, it does show it as a USB device using the device manager.

did anyone encounter it and know if there is a way to fix it so that the android will show it as a camera in the camera manager?

juice77ooO, 2023-07-18:
> Hey, I'm trying to create an android app that previews the purethermal image, And when I try to enumerate the connected cameras in android using Camera manager it doesn't list the purethermal. [...]

hi,

do you solve it? I have the same problem.

siamaktavakoli, 2023-10-22:
Hi, is this issue of yours fixed?