groupgets/purethermal1-firmware #58: Dissorted Picture while reducing bandwidth

snarayanan3082, 2023-09-20:
I am trying to use PureThermalMini (fw: v1.3.0) with i.MX8M Mini Development Board. At initial, i got 
> uvcvideo 1-1:1.1: Failed to submit URB 0 (-28)

So i changed the `VIDEO_PACKET_SIZE_ALT1` to `482`. Then i am getting a distorted image.
[image]

Any idea what to get rid of that line?

afzalhasan123, 2025-03-20:
Hey @snarayanan3082  Can you please help how did you do this. As I am also trying to modify but seems like nothing happen after updating the firmware.

I am also using Purethermal Lepton3.5 with i.MX8M Mini Development Board.

 Here is the path I used to change the packet size:
Middlewares/ST/STM32_USB_Device_Library/Class/video/Inc/usbd_uvc.h

-#define VIDEO_PACKET_SIZE                             ((unsigned int)(962))
+#define VIDEO_PACKET_SIZE                             ((unsigned int)(482)) /* min 162 for L2 Y16, 482 for L3 Y16 */
 
After applying changes and try to capture and stream but nothing works again same error:  Failed to submit URB 0 (-28)

snarayanan3082, 2025-03-20:
Hi @afzalhasan123 Can you confirm your firmware version also.
And also, I have done it long back. Currently I don't have any of the hardware with me. I have only made the changes i mentioned.

afzalhasan123, 2025-03-20:
My firmware version is 1.3.0 

[image]

afzalhasan123, 2025-03-21:
Hello @snarayanan3082 Now I am able to stream but same happens dissorted picture. Is your query solved regarding dissorted picture ?